snidercs/frc-bot-2025 #45: Python: Stress Test Timings

mfisher31, 2025-02-15:
... in other words force the latency down to 4ms and see if it breaks down.  The default, according to a fresh project, says the default in python is 20ms.

Our lua/c++ combo last year was running 4ms on the RIO and 1ms in the sim.  This issue is to serve as a data point in performance comparison of difference approaches in writing firmware.

The latency time mentioned above directly affects the responsiveness of GameController/Joystick to robot action.  It also directly affects the granularity of realtime parameter changes happening in hardware through code.

mfisher31, 2025-02-15:
As an aside, we might be able to elimate allocations happening in Python by creating object caches and updating things rapidly instead of `InstantiatingThemThroughAContstructor()`

martincarapia, 2025-02-25:
He figured this out during our stand up

PR #67 (2025-02-28): add: constant to tweak latency parameter
closes #45
closed 2025-02-28 by martincarapia