snidercs/frc-bot-2025 #49: Clean the RIO

mfisher31, 2025-02-22:
Just run this command: `robotpy undeploy` and see what happens.  The goal is to erase the things we were trying on 2/21/2025

martincarapia, 2025-02-23:
(base) richarddawkins@Mac pybot % robotpy undeploy
This will stop your robot code and delete it from the RoboRIO. Continue? [y/n]y
19:05:33:702 INFO    : robotpy.installer   : Settings loaded from /Users/richarddawkins/Documents/Code/ROBOTICS/frc-bot-2025/pybot/.wpilib/wpilib_preferences.json
19:05:33:704 INFO    : robotpy.installer   : Finding robot for team 9431
19:05:33:720 INFO    : robotpy.installer   : -> Robot is at 10.94.31.2
19:05:33:720 INFO    : robotpy.installer   : Connecting to robot via SSH at 10.94.31.2
19:05:33:932 INFO    : paramiko.transport  : Connected (version 2.0, client OpenSSH_8.3)
19:05:34:160 INFO    : paramiko.transport  : Auth banner: b'NI Linux Real-Time (run mode)\n\nLog in with your NI-Auth credentials.\n\n'
19:05:34:161 INFO    : paramiko.transport  : Authentication (password) successful!
SUCCESS: Files have been successfully wiped!
(base) richarddawkins@Mac pybot %

Output for that command. Code was redeployed Sat Feb 22nd with main.

closed 2025-02-23 by martincarapia

mfisher31, 2025-02-23:
perfect, thanks for doing that!  Looks like it "just works"