snidercs/frc-bot-2025 #52: Enable Field Centric Driving

mfisher31, 2025-02-22:
It is not enabled by default.  There was a "Left bumper" action to handle it but according to @crey-09, this does not work.  We tried to hard-code it, but that didn't work either.

mfisher31, 2025-02-22:
yep.  See `robotcontainer.py` and search for "field_centric"  you'll see what we did.  My next plan was to take that function call (`self.drivetrain.seed_field_centric()`) and repeatedly call it from inside `robotPeriodic()`.  Total guess, no idea what will happen :)

martincarapia, 2025-02-23:
Diagnoses:
Pigeon which is responsible for field centric driving was not being recognized by the firmware. 
I checked out the Pheonix Tuner X Software to see if the pigeon was being recognized by CAN, it wasn't. So, it wasn't a wrong ID in the firmware. 

After examining the robot closer, the pigeon wasn't giving "happy lights". Power wasn't being given to it. Code isn't at fault here.

closed 2025-02-23 by martincarapia

mfisher31, 2025-02-23:
> Diagnoses: Pigeon which is responsible for field centric driving was not being recognized by the firmware. I checked out the Pheonix Tuner X Software to see if the pigeon was being recognized by [...]

nicely done.