snidercs/frc-bot-2025 #81: Unit Tests In Python

mfisher31, 2025-03-03:
Whatever it takes to get unit tests running, do it.

mfisher31, 2025-03-03:
The test suite crashes around line 110 in telemetry.py.  There is a segmentation fault that gets triggered in this block of code.  Don't know why:

`telemetry.py`
```python

# Telemeterize the module states to a Mechanism2d
for i, module_state in enumerate(state.module_states):
    self._module_speeds[i].setAngle(module_state.angle.degrees())
    self._module_directions[i].setAngle(module_state.angle.degrees())
    self._module_speeds[i].setLength(module_state.speed / (2 * self._max_speed))

    SmartDashboard.putData(f"Module {i}", self._module_mechanisms[i])

```

mfisher31, 2025-03-03:
CTRE is aware of this crash: currently their recommendation is essentially "hope for the best":
https://www.chiefdelphi.com/t/fatal-python-error-segmentation-fault-when-using-register-telemetry-in-phoenix-6-swerve-drive-api/483721/4?u=mfisher31

JaydenGordan, 2025-03-11:
I'll look at it but I'm not to sure how much I can get it done since I won't have my personal laptop on me today

KawiMinn, 2025-03-12:
@mfisher31 did you do that with custom unit tests, or just the built-in ones? If it was just the built-in ones, all 4 are passing for me now, after a few changes. Not sure I fully understand how robotpy test works though, so there's probably some things I'm just not aware of, and thus missing.

mfisher31, 2025-03-13:
Interesting... I meant the built-in tests.  I wonder why these pass for you but not me (or any other *nix system)