snidercs/frc-bot-2025 #88: Auto v2 - Vision Support and Coral Delivery

mfisher31, 2025-03-06:
Use the Limelight to line up and deliver a coral on L2.  Path planner or choreo is not needed in this version since it will use vision tracking to move to and line up the shot.  This issue is dependent on smooth driving and general limelight integration to be in place.

mfisher31, 2025-03-06:
see also #83 : Its looking like we really will want this considering autos with limelight can't just load a file and run.

KawiMinn, 2025-03-09:
[image]

Limelight data can be found in network tables it seems

martincarapia, 2025-03-09:
https://docs.limelightvision.io/docs/docs-limelight/pipeline-apriltag/apriltag-robot-localization-megatag2

In 2024, most of the WPILib Ecosystem transitioned to a single-origin coordinate system. In 2023, your coordinate system origin changed based on your alliance color.

For 2024 and beyond, the origin of your coordinate system should always be the "blue" origin. FRC teams should always use botpose_orb_wpiblue for pose-related functionality