snidercs/frc-bot-2025 #99: RPi3 or RPi4 for Dumb Camera

mfisher31, 2025-03-09:
Adding this as a 'note to self'.  If by chance the RoboRIO can't handle processing the video feed, we should offload to a rpi3 or 4.   The camera is only for video display so there wouldn't be any need for  Pi <-> RIO communications.

@mfisher31 has a raspberry pi 3b+ that could be used for this purpose.

martincarapia, 2025-03-09:
CPU usage hovers around 90% so maybe? We'll have to do some driving to see if it really affects anything....
MINN PUT SCREENSHOTHERE

KawiMinn, 2025-03-09:
[image]

mfisher31, 2025-03-09:
90 is pretty high.  And when the code starts generating trajectories that is likely to try and go higher because of all the heap allocations that could rapidly start taking place.

martincarapia, 2025-03-09:
Limelights actually have been found to do the grunt work for us. It'll tell us our position on the field based on the april tag that it sees, and also give us values to go to the april tag. I'll have to show you for it to make snese, but we're able to grab that data. Although, still being at 90 percent is pretty high. If we could offload it to a seperate processor it'd destress our rio.

martincarapia, 2025-03-11:
@KawiMinn Could you research how you would flash the raspberry pi and set up the software so that the camera feed can be shared over NetworkTables?

martincarapia, 2025-03-14:
Got it figured out. Closing. CPU usage down by 20% and will continue with efficiency updates.